sy2002/MiSTer2MEGA65 #5: Show the input CORE screen resolution

MJoergen, 2023-05-14:
The M2M framework now automatically detects the screen resolution (i.e. visible pixels) received from the CORE. Inside the "monitor" this can be retrieved as follows:
```
QMON> MEMORY/CHANGE ADDRESS=FFF4 CURRENT VALUE=0101 NEW VALUE=00FF
QMON> MEMORY/CHANGE ADDRESS=FFF5 CURRENT VALUE=0001 NEW VALUE=0010
QMON> MEMORY/DUMP START ADDRESS=7003 END ADDRESS=7004

7003: 0180 010E 
```
The above is from the C64MEGA65 core, and shows that the horizontal size is 0x0180 = 384 pixels, while the vertical size is 0x010E = 270 lines.

This information should also be available without having to enter the "monitor". E.g. either displayed to UART upon start-up, and/or visible somewhere in the OSM.

MJoergen, 2023-05-23:
Please also add the following:

- Horizontal sync frequency (i.e. number of scan lines per second). This is read off directly as `@0x7009`.
- Vertical sync frequency (i.e. number of frames per second). This has to be calculated as follows:`@0x7009 / @0x7008`. Should preferably be rounded to one decimal (as opposed to just displaying an integer).

sy2002, 2023-05-23:
> Should preferably be rounded to one decimal

@MJoergen As you are the math wizard: QNICE does not have any floating point maths. Can you help me with a simple floating point algorithm either written in C or in QNICE assembly (latter one preferred since a lot is on my plate for the release)? These are the operations that we have: 32bit multiplication, 32bit division, 3bit addition, 32bit subtraction.

MJoergen, 2023-05-23:
Just for the record, these are all the variables provided:

- 0x7003 : Horizontal visible pixels (calculated by ascaler)
- 0x7004 : Vertical visible pixels (calculated by ascaler)
- 0x7005 : Horizontal visible pixels (expected to be same as 0x7003)
- 0x7006 : Horizontal total pixels per scanline (including blanking area).
- 0x7007 : Vertical visible lines (expected to be same as 0x7004)
- 0x7008 : Vertical total lines per frame (including blanking area).
- 0x7009 : Horizontal total lines per second.

MJoergen, 2023-05-23:
> @MJoergen As you are the math wizard: QNICE does not have any floating point maths. Can you help me with a simple floating point algorithm either written in C or in QNICE assembly (latter one [...]

Ok, then I suggest we for now just forget the floating point and just print out the result of integer division.

For some later project, I suggest we add a more general floating point library to QNICE.

For the record, one algorithm to achieve the requested "division rounded to one decimal" is the following:

- Goal: Display `x/y` rounded to 1 decimal place.
- Step 1 : Perform the integer division `(10*x) / y`. Call this number `z`.
- Step 2 : Calculate the integer division `z/10` and display it. This is the integer part of the result.
- Step 3 : Calculate `z - 10*(z/10)` and display it. This is a digit from 0 to 9, and is the first decimal digit.

MJoergen, 2023-05-30:
I've changed the register yet again, so that they now match the parameters found in `M2M/vhdl/av_pipeline/video_modes_pkg.vhd`:

- 0x000 : CORE_X (determined by ascaler; should be equal to H_PIXELS)
- 0x001 : CORE_Y (determined by ascaler; should be equal to V_PIXELS)
- 0x002 : H_PIXELS
- 0x003 : V_PIXELS
- 0x004 : H_PULSE
- 0x005 : H_BP
- 0x006 : H_FP
- 0x007 : V_PULSE
- 0x008 : V_BP
- 0x009 : V_FP
- 0x00A : H_FREQ

I suggest you just display these raw values in decimal, without any further post-processing.

For the user it is relevant to be able to determine the following derived quantities:

- Frame rate (in Hz) : `H_FREQ / (V_PIXELS + V_PULSE + V_BP + V_FP)`
- Pixel rate (in Hz) : `H_FREQ * (H_PIXELS + H_PULSE + H_BP + H_FP)`

MJoergen, 2023-05-30:
Perhaps displaying `V_TOTAL = V_PIXELS + V_PULSE + V_BP + V_FP` and `H_TOTAL = H_PIXELS + H_PULSE + H_BP + H_FP` would be nice for the user.

sy2002, 2023-06-02:
Done with this commit: 98f4fe8c1505bbace1501bb0a3c9608492c96188

closed 2023-06-02 by sy2002