sy2002/MiSTer2MEGA65 #9: Add 60 Hz mode for 576p modes

sy2002, 2023-05-22:
Right now, the M2M framework supports these combinations of resolutions, aspect ratios and display frequencies:

```
16:9 720p 50 Hz = 1,280 x 720 pixel
16:9 720p 60 Hz = 1,280 x 720 pixel
4:3  576p 50 Hz =   720 x 576 pixel
5:4  576p 50 Hz =   720 x 576 pixel
```

This is great for the C64 in PAL mode. But for a future version of the C64 that also supports NTSC, as well as for core which are running at 60 Hz such as Galaga, we need to support more of these combinations. My suggestion is that we add:

```
4:3  576p 60 Hz =   720 x 576 pixel
5:4  576p 60 Hz =   720 x 576 pixel
```

There are two approaches to this:

1) Can ASCAL generate the 576p @ 60 Hz without the need for additional clocks? The 720p 50 Hz and 60 Hz are generated from the same clock. ASCAL just again does its "magic". Approach one sounds simple and elegant.

2) Another approach is working with more clocks - which is a chore and not elegant at all. Maybe this would then be the final "straw that breaks the Camel's back" that motivates us to switch to fully dynamically generated clocks using the reconfiguration properties of the MMCM and the mechanisms that Adam Barnes implemented in the tyto2 project? He can seamlessly flip through 16 video modes: https://github.com/amb5l/tyto2/blob/main/src/designs/hdmi_tpg/mega65r3/hdmi_tpg_mega65r3.vhd

No matter what route: I propose not to touch anyting here before we released V5 of the C64 core.

MJoergen, 2023-05-23:
In the following, I'm referring to the document CEA-861-D, which can be found [here](https://ia803002.us.archive.org/1/items/CEA-861-D/CEA-861-D.pdf).

Currently, we are supporting the following HDMI formats:

- 720p @ 60 Hz (section 4.3, format 4, pixel clock 74.25 MHz, aspect ratio 16:9)
- 720p @ 50 Hz (section 4.7, format 19, pixel clock 74.25 MHz, aspect ratio 16:9)
- 576p @ 50 Hz (section 4.9, format 17, pixel clock 27.00 MHz, aspect ratio 5:4)

I don't see any 576p @ 60 Hz formats mentioned in this standard.

However, I do see these two alternatives:

- 480p @ 60 Hz (section 4.2, format 1, pixel clock 25.2 MHz, aspect ratio 4:3)
- 480p @ 60 Hz (section 4.5, format 2, pixel clock 27.027 MHz, aspect ratio 3:2)

Moving forward, I suggest we consider the first alternative (aspect ratio 4:3). In any case, either choice requires the introduction of yet another clock. (Note: I'm pretty sure that the closeness of 27.00 MHz and 27.027 MHz is an illusion and trying to exploit this will cause problems on picky HDMI monitors.)

However, I don't think an additional clock is a great chore. To me it seems like a rather trivial matter of updating the two files: `M2M/vhdl/av_pipeline/video_modes_pkg.vhd` and `M2M/vhdl/clk_m2m.vhd`. All the magic is done in the latter file, based on the clock select input `hdmi_clk_sel_i`.

Switching to a "fully dynamically generated clocks" I see - at the moment - as a nice-to-have, i.e. a refactoring that can be implemented when time permits, but is not urgent.

sy2002, 2023-05-23:
Thank you for the insights. Let's then please move merging this to the M2M framework to after the C64 V5's release (just in case you plan to tackle this any time soon). Right now this is only relevant for Galaga.

sy2002, 2023-06-17:
Also relevant for the GameBoy.

sy2002, 2023-11-02:
**Status quo on November 2, 2023:**

@MJoergen has added Adam Barne's dynamic reconfiguration feature to the framework which enables us to easily switch between different resolutions. Questions around "59.94 vs. 60 Hz" remain. There is not a lot of clarity or know how on my (sy2002's) side yet, what the right path forward is. So we need to research and decide.

Here is a chat from Skype from sy2002 to MJoergen:

* 59.94 vs 60 Hz: Adam's Excel sheet that is the input of his python script that generates the different re-programming sequences says that he is doing 59.94 which is "correct NTSC" as far as I know. I am not very savvy, when it comes to these American TV standards. It is cool that PAL was simply 50 Hz and done.
* I am aware though, that certain cores - such as GameBoy - output EXACTLY 60 Hz, because the GameBoy's display was an LCD display and not a TV. So to avoid that the GameBoy core flickers when side-scrolling (just like Giana Sisters did before we had the HDMI flicker fix), I think we do need to support additionally also exactly 60 Hz.
* CPP needs to have a choice. Other cores, such as Arcades output NTSC-ish 59.94. So for them, 60 Hz would be wrong.
* Also the document that we work with (CEA861D) seems to distinguish. I am not sure if it is enough to use a different pixel clock to distinguish between 59.94 or if you use other front porch, back porch and stuff. Can you please research this?
* MiSTEr itself only seems to support clean 60 Hz. I am not sure what decision is right. 

Links to MiSTer:

MiSTer's Video Modes: https://mister-devel.github.io/MkDocs_MiSTer/advanced/videomodes/
Also interesting they use this "ADV chip" that the MEGA R2 used.

Native video modes and frequencies of MiSTer cores:
https://mister-devel.github.io/MkDocs_MiSTer/advanced/nativeres/

MJoergen, 2023-11-02:
Regarding difference between frame rates 59.94 Hz and 60.00 Hz, this is determined by the Pixel Clock frequency. The following is a summary from the CEA861D document:

For 59.94 Hz we have the following choices:

- 640x480 : 25.175 MHz
- 720x480 : **27.000 MHz**
- 1280x720 : 74.176 MHz

For 60.00 Hz we have the following choices:

- 640x480 : **25.200 MHz**
- 720x480 : 27.027 MHz
- 1280x720 : **74.250 MHz**

I've highlighted in **bold** the pixel frequencies currently supported in the framework.

MJoergen, 2023-11-02:
The following comment is copied from here: https://obsproject.com/forum/threads/59-94-vs-60-29-97-vs-30-fps.101408/

> If you're talking about matching exact frame rates, then any time you're dealing with true NTSC frames it's going to be either 29.97 or 59.94. This is true for broadcast television, VHS, DVD, anything that outputs a composite, s-video, or component signal.
>
> HDMI is where things start to change. Most home consoles still use the NTSC standard of 59.94fps, but this is actually just a holdover. There's no requirement for HDMI to adhere to the NTSC standard, and this is very much true for computer monitors where you're likely getting a flat 60fps display now.
>
> For capture, as R1CH said, the only time you should ever really worry about exactly matching framerate is when dealing with retro sources, specifically for the purpose of deinterlacing. As long as the deinterlacing algorithm is behaving nicely, all it needs is to be fed the exact top/bottom frame order and it will be able to output a smooth picture. If there's any desync though (say, from assuming there's supposed to be an extra frame because it's expecting 30fps instead of 29.97fps), that will result in deinterlacing artifacts until it can resync -- lines will look doubled, or be bobbing rapidly, or it will just look interlaced still.
>
> For progressive capture, the impact of a lost frame due to desync is just that -- a lost frame.
>
> To put things in perspective a little bit more, the difference between 59.94fps and 60fps is 1 frame every 16.7 seconds. For the purposes of capture, as long as you're feeding a progressive scan, it honestly doesn't matter if you chose to go with 59.94 or 60 for your output framerate. Most likely, random framestutters from the capture chain, render hiccups, or even playback stutter are going to be much more noticeable.
>
> Another thing to keep in mind. Most playback devices that people are watching on are now 60fps devices -- monitors, phones, even TVs. There is going to be a framerate conversion at some point in the chain if this is the case -- all that changes is where that framerate conversion happens.

sy2002, 2023-11-02:
Thank you for your research @MJoergen : I would vote for giving the Core Porting Person (CPP) the choice between 59.94 Hz or 60 Hz just because otherwise we might very well have a stuttering in for example horizontal scrolling for retro machines that do "real" NTSC.

MJoergen, 2023-11-03:
The missing clocks can be generated using these MMCM settings:

 Desired | Actual | `CLKFBOUT_MULT_F` | `DIVCLK_DIVIDE` | `CLKOUT0_DIVIDE_F`
 ------- | ------ | ----------------- | --------------- | ------------------
 25.175  | 25.174 |  27.125           | 2               | 53.875
 74.176  | 74.176 |  50.625           | 7               |  9.750
 27.027  | 27.027 |  10.000           | 1               | 37.000

MJoergen, 2023-11-03:
Using xapp888 I get the following DRP values:

## 25.175
```
 06 1145
 07 0000
 08 10c4
 09 0080
 0A 1452
 0B 0080
 0C 1145
 0D 0000
 0E 1145
 0F 0000
 10 1145
 11 0000
 12 1145
 13 3000
 14 1208
 15 5800
 16 0041
 18 013f
 19 7c01
 1A 7de9
 28 FFFF
 4E 9900
 4F 1100
```

## 74.176
```
 06 1145
 07 0000
 08 1041
 09 0000
 0A 1145
 0B 0000
 0C 1145
 0D 0000
 0E 1145
 0F 0000
 10 1145
 11 0000
 12 1145
 13 0c00
 14 128a
 15 2c00
 16 0042
 18 00c2
 19 7c01
 1A 7de9
 28 FFFF
 4E 1100
 4F 9000
```

## 27.027
```
 06 1145
 07 0000
 08 1104
 09 0000
 0A 1514
 0B 0000
 0C 1145
 0D 0000
 0E 1145
 0F 0000
 10 1145
 11 0000
 12 1145
 13 3000
 14 128a
 15 5800
 16 0041
 18 00db
 19 7c01
 1A 7de9
 28 FFFF
 4E 9000
 4F 0100
```

MJoergen, 2023-11-04:
With the latest commit (b0e97d6) we now support the following modes:

- [x] 720p 50 Hz 16:9
- [x] 720p 60 Hz 16:9
- [x] 576p 50 Hz 4:3
- [x] 576p 50 Hz 5:4
- [x] 640x480 60 Hz
- [ ] 720x480 60 Hz
- [ ] 640x480 59.94 Hz
- [x] 720x480 59.94 Hz
- [ ] 720p 59.94 Hz

I've verified (by measurement) that the video clock frequency is correct for all nine modes. However, my monitor (Philips S-line 275S1AE) only displays some of the modes (marked with a check mark above)

sy2002, 2023-11-04:
So here are the resolutions, that we should IMHO keep because they seem to be official and work (with the exception of 800x600):

720p @ 50 Hz 16:9
720p @ 60 Hz 16:9
576p @ 50 Hz 4:3
576p @ 50 Hz 5:4
640 x 480 @ 60 Hz
720 x 480 @ 59.94 Hz
800 x 600 @ 60 Hz  (by the way it is not called 600p, it is called 800 x 600).

What does not work on your neither on mine monitor is:
720 x 480 @ 60 Hz
640 x 480 @ 59.94 Hz

The 720p @ 59.94 Hz is recogized 2562 x 720 and flickers as nasty as the 720p @ 60.

MJoergen, 2023-11-05:
With the latest commit (a40dc94) we now support the following modes:

- [x] 720p 50 Hz 16:9
- [x] 720p 60 Hz 16:9
- [x] 576p 50 Hz 4:3
- [x] 576p 50 Hz 5:4
- [x] 640x480 60 Hz
- [x] 720x480 59.94 Hz
- [x] 800x600 60 Hz

MJoergen, 2023-11-05:
I'm closing now, since everything is implemented and tested on my monitor.

closed 2023-11-05 by MJoergen