sy2002/MiSTer2MEGA65 #17: Sprites not shown correctly

Schwefelholz, 2023-06-09:
I've got a safety copy of "Raid on Bungeling Bay" that shows a "ghost image" of the helicopter's rotor Blade.
[image]

The same is shown correctly on an Ultimate 64
[image]

sy2002, 2023-06-09:
@Schwefelholz Thank you for reporting. You chose the repository "MiSTer2MEGA65", which is the wrong repository, as "MiSTer2MEGA65" is a framework that allows programmers to port cores from the MiSTer retro computing project to the MEGA65 project.

Googling the game name you mentioned, I assume you are talking about a C64 game.

Can you quickly clarify, if you observed this bug with the **MEGA65 ("go 64")** mode of the C64?

Or, did you observe this bug with the **dedicated C64 for MEGA65 core**? If so, with which version?

I am happy to move/transfer your bug report into the right repository which would either be:

https://github.com/MEGA65/mega65-core   in case of MEGA65 "go 64"    or

https://github.com/MJoergen/C64MEGA65  in case of the dedicated C64 for MEGA65 core.

I am closing this issue as invalid as this is the wrong repo - but as said: Please give me feedback and we take it from there.

Schwefelholz, 2023-06-09:
Hi @sy2002.
Sorry für having chosen the wrong repository.
I observed this issue on the C64MEGA65 core.

sy2002, 2023-06-09:
Thank you @Schwefelholz I will transfer the issue to the C64MEGA65 repo.

sy2002, 2023-06-09:
@Schwefelholz OK, done: It is now tracked in the C64MEGA65 repo: https://github.com/MJoergen/C64MEGA65/issues/76