sy2002/MiSTer2MEGA65 #20: Supply up a list of ported cores

hansfbaier, 2023-07-25:
Hello, porting MiSTer cores to Xilinx is also very important for MiSTeX,
a MiSTer port to different FPGA boards, including, but not exclusive to Xilinx FPGAs.

I hope we can communicate.
I already have a working port for the QBert arcade core.

https://github.com/MiSTeX-devel

MJoergen, 2023-07-25:
That sounds interesting!

This project originally grew out of porting the GameBoy MiSTer core to the MEGA65, but the GB core is not based on this M2M framework. When we started porting the C64 MiSTer we realized that a generic framework would make future porting more easy, so that is the first core to use the M2M framework. The M2M framework is loosely based on the work done porting the GB core, but has since evolved considerably. Currently, we're working on the Galaga and Apple II MiSTer cores as our next use-cases. We would of course like to modify the GameBoy port to use the M2M framework as well. This should not be too big a task.

The above paragraph should probably be reworded and added to the README file.

My first priority with this repo is to update the documentation and the tutorials, because that has been lacking for a long time.

Next, we will probably "attack" the Amiga500 core. This will probably require modifications to the framework.

We have the challenge of supporting different revisions of the MEGA65 hardware (some with and some without extra SDRAM). But you seem to have an ever larger challenge supporting a wide range of boards, if I understand you correctly.

Do you have a list of the different platforms you're supporting, and their different capabilities/features ?

hansfbaier, 2023-07-25:
Yes, I use the LiteX SoC generation framework to define the different boards.
https://github.com/MiSTeX-devel/MiSTeX-boards/tree/main/mistex_boards

hansfbaier, 2023-07-25:
I still have some trouble porting block RAM to Xilinx, some of the cores I ported just don't work there (yet).

closed 2023-10-17 by sy2002