sy2002/MiSTer2MEGA65 #21: Guarantee correst placement constraints

sy2002, 2023-10-20:
The pin layout on R4 and R5 is pretty different from R3/R3A. Therefore:

From a Skype message from @sy2002 to @MJoergen:

The placement contraints in the XDC for HyperRAM, keyboard, SD card controller and VGA output registers where mission critical to fix certain "odd bugs" that we encountered. You can walk through the closed issues on GitHub in the C64 folder. So we need to take this seriously. Since the PIN positions changed: Did you ensure that the new position is in line with what is described in the comment?

MJoergen, 2023-10-23:
I've verified pin placement of the R4 board, and they match the existing Pblocks. So nothing more needs done there.

I'm leaving this open for the R5 board.

MJoergen, 2023-11-05:
I've now verified that the pin placement of critical blocks (HyperRAM, SD Card, etc) are unchanged between R4 and R5. I'm therefore closing this issue.

closed 2023-11-05 by MJoergen