sy2002/MiSTer2MEGA65 #23: HDMI compliancy: Assert +5V according to section 4.2.7 of the specification version 1.4b

sy2002, 2023-10-20:
HDMI compliancy: Assert +5V according to section 4.2.7 of the specification version 1.4b

This very line of code was a game changer when it comes to HDMI compatibility:

```
   ct_hpd_o           : out   std_logic := '1';
```

It is here in the R3 version of the core:

https://github.com/MJoergen/C64MEGA65/blob/develop/M2M/vhdl/top_mega65-r3.vhd#L43C72-L43C160

TODO:

-[ ] Add this to the R4 board implementation
-[ ] Have a test user confirm that it works (either by measuring the +5V or by using one of the devices, for example HDMI capture device, that do not work without these +5V)
-[ ] Add this to the R5 board implementation
-[ ] Have a test user confirm that it works

Why "confirm that it works?": Because the R4 and R5 boards a new and might have bugs on the board

@MJoergen You can assign this to me after you fixed it in the code for the "confirm" part of the job: I know the users and the issue quite well.

sy2002, 2023-10-21:
@MJoergen Is this comment that you made here: https://github.com/MJoergen/C64MEGA65/issues/82#issuecomment-1773813754 related to this? If yes, then we could check the boxes "Add this to the board implementation" and only have people test it - or - if we are sure that this power-supply you are talking about is really hardwired on the board, we can also just close this one.

MJoergen, 2023-10-22:
Yes, this is indeed related. I'm pretty sure it just "works out of the box", but it would be nice to have someone confirm it.

sy2002, 2024-01-29:
Deft confirmed that the C64 core V5.1 RC1 on an R5 board works fine with his HDMI capture device.

closed 2024-01-29 by sy2002