sy2002/MiSTer2MEGA65 #34: CRT/ROM loading protocol: Less copy/pasting, less hardcoding

sy2002, 2023-11-12:
We agreed upon a protocol between the firmware and "CRT or ROM loading hardware". Examples which are using this protocol currently are in the C64 core:

* Software `.crt` loading
* Program file `.prg` loader

The protocol consists of a few state machines and a few constants.

Right now the state machines as well as the constants are always copy/pasted into the multiple "clients/servers" of this protocol. 

1) The minimum we should do is to put the constants into an official M2M package and then the CPP would import the package instead of copy/pasting the constants all the time.

2) Optional for bonus points: Can we write some generic code that CPPs can use to implement this protocol (and if so, we would then refactor the above-mentioned "users").

```vhdl
   -- Status reporting from QNICE/firmware:
   constant C_CRT_ST_IDLE         : std_logic_vector(15 downto 0) := X"0000";
   constant C_CRT_ST_LDNG         : std_logic_vector(15 downto 0) := X"0001";
   constant C_CRT_ST_ERR          : std_logic_vector(15 downto 0) := X"0002";
   constant C_CRT_ST_OK           : std_logic_vector(15 downto 0) := X"0003";

   -- Status reporting to QNICE/firmware
   constant C_STAT_IDLE           : std_logic_vector(3 downto 0) := "0000";
   constant C_STAT_PARSING        : std_logic_vector(3 downto 0) := "0001";
   constant C_STAT_READY          : std_logic_vector(3 downto 0) := "0010"; -- Successfully parsed CRT file
   constant C_STAT_ERROR          : std_logic_vector(3 downto 0) := "0011"; -- Error parsing CRT file

   -- Control and Status register
   constant C_CRT_CASREG          : unsigned(15 downto 0) := X"FFFF";
   constant C_CRT_STATUS          : unsigned(11 downto 0) := X"000";
   constant C_CRT_FS_LO           : unsigned(11 downto 0) := X"001";
   constant C_CRT_FS_HI           : unsigned(11 downto 0) := X"002";
   constant C_CRT_PARSEST         : unsigned(11 downto 0) := X"010";
   constant C_CRT_PARSEE1         : unsigned(11 downto 0) := X"011";
   constant C_CRT_ADDR_LO         : unsigned(11 downto 0) := X"012";
   constant C_CRT_ADDR_HI         : unsigned(11 downto 0) := X"013";
   constant C_CRT_ERR_START       : unsigned(11 downto 0) := X"100";
   constant C_CRT_ERR_END         : unsigned(11 downto 0) := X"1FF";
```

MJoergen, 2023-11-19:
I like this idea.

I've made a commit in the C64 repo (because it's easier to test there) : https://github.com/MJoergen/C64MEGA65/commit/92921e6bb36391f4308e801ac6bca39158988a53

I've refactored the CRT loading, but refactoring the PRG loading is still TBD.

@sy2002 : Please review.

MJoergen, 2023-11-19:
I'll do the PRG loading too.

MJoergen, 2023-12-20:
prg_loader refactored in C64 branch "mfj_refactor_prg". Not yet tested.

sy2002, 2024-01-05:
@MJoergen I tested the refactored PRG loader with multiple games from the "one load collection": Works like a charme. Do be on the safe side I also tested multiple `.crt` cartridges: Work like a charme, too. I am closing this issue.

closed 2024-01-05 by sy2002