sy2002/MiSTer2MEGA65 #35: Board-dependent HyperRAM phases

sy2002, 2023-11-15:
Labeled as V2.0.0 and needed for the C64 core V5.1:

We found that each board revision seems to need its own HyperRAM phase. We need to experimentally confirm the right phases for R4 and R5 and we stick to the known-good phase for R3/R3A.

MJoergen, 2023-11-17:
Fixed in commit d44a3a3

closed 2023-11-17 by MJoergen

sy2002, 2023-11-17:
@MJoergen Were we already done finding a stable Phase for R4? Curious to learn more about your thought process.

MJoergen, 2023-11-17:
The ability to have a constant phase shift different for each board revision is now implemented.

The actual value chosen for R4 (and R5) is a "good enough" value based on the recent feedback in the Skype channel.

However, I'm planning on doing a more systematic test, using this branch: https://github.com/MJoergen/HyperRAM/tree/mfj_phase_shift