sy2002/MiSTer2MEGA65 #36: Research 200 MHz HyperRAM for R4/R5 boards

sy2002, 2023-11-15:
Looks like the newer HyperRAM chips on R4/R5 boards are able to cope with 200 MHz instead of the 100 MHz we are currently using. Would be a very welcome latency reduction and speed boost.

MJoergen, 2023-11-19:
In the HyperRAM repository I've created a branch "mfj_175MHz". Link: https://github.com/MJoergen/HyperRAM/tree/mfj_175MHz

MJoergen, 2024-02-08:
I'm not sure where this idea about a 200 MHz HyperRAM came from, and I think it's false.

The R3 board uses a `IS66WVH8M8BLL-100B1LI` chip (for U29), which indeed is rated to 100 MHz.

However, the R5 board uses a `IS66WVH8M8DBLL-100B1LI` chip (for U29), which is identical except it is "die revision D". Specifically, this chip is also only rated to 100 MHz.

MJoergen, 2024-02-08:
I'm guessing the confusion arose because the revision D does indeed come in a 200 MHz version, but the R5/R6 boards only use the 100 MHz version.

MJoergen, 2024-02-08:
However, what is confusing me is that section 10.3 of the revision D document above only details timing values for 200 MHz, 166 MHz, and 133 MHz versions. So timing details for the 100 MHz is not even documented!

sy2002, 2024-02-08:
@MJoergen Well, then we might want to close this "research" project here - shouldn't we?

closed 2024-02-12 by MJoergen