sy2002/MiSTer2MEGA65 #41: Add remote control of the core (via UART and/or Ethernet)

MJoergen, 2024-03-18:
The use case is to be able to control the core without needing physical access to the MEGA65. This could either be done over the serial (UART) port or over Ethernet.

This feature can be implemented in several stages of progressing usability and complexity.
1. The first stage is to forward serial input to the core's keyboard input.
2. The second stage is to copy the OnScreen Menu frame buffer to the serial output (as text only).
3. The video output from the core could be compressed and sent over the Ethernet, one frame at a time per request. In other words, the user types a keyboard sequence to force a screenshot (snapshot) of the current output frame. The protocol/format would be something proprietary, but easy to implement in the FPGA. This would be decompressed and converted to an image on the host computer.
Note: I already have a proof-of-concept developed for another project on the Nexys4DDR board (https://github.com/MJoergen/nexys4ddr/tree/master/c64).
4. This stage would continuously generate a stream of data sent over the Ethernet port. This would be converted to a video file, to be viewed later (not real time). 
5. The final stage would implement protocol support for some kind of Remote Desktop, that can connect to a standard client on the host machine. This requires a complete IP stack with TCP and UDP support

I expect stages 1-4 to be do-able with very modest resource requirements in the FPGA, and also modest development resources, given that there is already a working POC.
Stage 5 is much more complex: TCP is usually implemented in software, not hardware, so this will likely involve the QNICE CPU, perhaps only after interrupt support and/or DMA support is implemented.

gurcei, 2024-03-27:
This does sound cool in the sense of remote controlling menus and remote desktopping in, though part of me wonders about the ability to step debug issues on the c64 via a uart monitor too. If it is of a similar nature (same commands) to the mega65's uart monitor, then it could mean we could make use of existing tooling like m65dbg too. But anyways, I won't get in the way of your immediate plans if the more exciting goal is the remote control/desktop experience 🙂

MJoergen, 2024-03-27:
I've found the following documentation of m65dbg:

https://github.com/MEGA65/mega65-core/blob/development/docs/usermanual0.md#43-remote-serial-monitor-handy-for-debugging

https://github.com/MEGA65/mega65-core/blob/development/docs/monitor.md

gurcei, 2024-03-27:
Some additional thoughts on existing uart monitor implementation within mega65-core in this discord thread:

https://discord.com/channels/719326990221574164/1222391068453244959/1222618515593494712