sy2002/MiSTer2MEGA65 #52: XOR 0, R9 bug in shell.asm?

sy2002, 2024-09-07:
@MJoergen found this:

[image]

I answered this:

[image]
[image]

We are talking about this:

https://github.com/sy2002/MiSTer2MEGA65/blob/V2.0.0-release/M2M/rom/shell.asm#L1047

Rhialto, 2025-02-16:
Ah, I noticed as well that the write enable for the vdrive track buffer remains on all the time, but even the 4040/8250 I'm importing right now ANDs this with the ACK signal.
https://github.com/eriks5/CBM-II_MiSTer/blob/master/rtl/ieee_drive/ieeedrv_drv.sv#L290
But it would be nicer to turn off the write enable.

Rhialto, 2025-07-01:
Of course! For the `develop` branch I expect? And I'll do it with `XOR R9,R9` to make it a bit shorter.

I can think of some further improvements, from simple to complicated:

1. Remove the in-effective `XOR     0, R9` / `RSUB    VD_CAD_WRITE, 1` from the loop to make iterations slightly faster.
2. Remove the code to turn on the write enable to before the loop, again to make iterations slightly faster. It would also need to set up the first address and data correctly, to prevent more messy garbage writes
3. Create some vhdl support for these loops,  to make it even super-fast.

The Mister is apparently a lot faster than we are, in servicing the "sd card" requests. Certainly in the case of my 8250, where I need to access hyperram. This long time caused bugs in the drive which wasn't expecting this. So I will take any easy speed-up here.

sy2002, 2025-07-01:
@Rhialto Yes, `develop` branch, please.

In general: The slow speed in our SD card handling is due to a very sub-optimal way how the underlying QNICE SoC handles SD cards in terms of how it buffers things and also when it comes to the actual SD card controller that QNICE uses. So please do not expect any noticable speedup by the optimizations you propose above. When preparing for an upcoming V6 release of the C64 core, we should refactor and optimize a lot of this underlying infrastructure. Otherwise I also see it coming, that for example for an Amiga or Atari ST core, the VDrive system will simply be too slow.

But still, we are doing this to fix a bug. So thank you for the patch :-)

We might also want to think about removing the in-effective "strobing" - or - do a real strobing, i.e. fixing the XOR bug. This code is rather old, so I need to wrap my head around everything in due time.

sy2002, 2025-07-02:
@paich64 can you please work with Rhialto to write a small Basic program and test this?

paich64, 2025-07-03:
@guys, I'm more than happy to help, but honestly i don't know what we are talking about here :D 
What is this Assembly code ? it does not sound like 6510 assembly.
What are you exactly expecting from me ? not sure i get what you really want to test :P

Rhialto, 2025-07-04:
@paich64 and others who didn't dive so deeply into this, I can give some context that hopefully explains.

The 1541 inside the C64 core (and in general floppy disk drives in MiSTeR cores) have some internal memory which generally stores a single track of data from the inserted floppy disk, so that it can access this data quickly. Let's call this the "track buffer".

Now if a new/different track is accessed, the track buffer needs to be saved back to the floppy image (if it was changed) and the data for the new track needs to be loaded into it.

For this purpose, there is the "sd card protocol" which the drive speaks, and it is understood by the program that is running on the QNice CPU which is implemented in the M2M framework. This code responds to the requests from the floppy and either writes the requested data into the track buffer, or reads it out and copies it into the floppy image, where it eventually ends up on the sd card.

The OQ (original question) of this issue is about that code, in particular about the code which writes *into* the track buffer. Effectively, it leaves the "write enable" signal turned on when it is done. At first it was thought this was harmless, because the floppy drives **and** this signal with the "acknowledge" signal, which is properly turned off after this. If the acknowledge signal is off, no data is written into the track buffer.

However, I discovered that it can have a bad effect after all. At least in the 8250 drive implementation I saw this in the form of "25, write error"s. When the QNice code services the other kind of request, to read data from the track buffer, it also uses the acknowledge signal at the end to tell the floppy that it has finished. And that makes both signals on again. A spurious byte is written into the track buffer.

Where? The last byte of the track is still addressed via the address lines.

Which value? I suspect(!) whatever value was last put to the "write byte data" signals. Which value that is? I suspect(!) it is the last value that was legitimately written into the track buffer, which would (I suspect!) also be the value of its last byte.

Effectively, if you changed the last sector, this bug will change the last byte of the last sector back to what it was.

If you change another sector, you probably don't notice.

Note that this corruption is, so far, only present in the track buffer. You could see it if you read the last sector of the track, but it won't be in the disk image.

To trigger that the corruption makes it into the disk image, you would need to make a second change of the track, so that the track buffer is for a second time read out and saved into the disk image. The bug of the corruption of the track buffer would happen again... but would it write the same byte value again? (probably yes, but it would be good to test).

So, summarizing: the bug is set up when data is written into the track buffer (because the C64 wants to read something from disk). Then it triggers every time data is read back from the track buffer (which is triggered when the C64 wrote something to disk). This makes that a program to verify if the bug also affects the 1541 has to do just the right things to test it properly.

Rhialto, 2025-07-06:
I had a quick look at https://github.com/MJoergen/C64_MiSTerMEGA65/blob/5afb90e02fcefb053ac27792f42e742ad61e8b9b/rtl/iec_drive/c1541_drv.sv#L246-L252 the 1541 code in the C64 core.

It looks like the "save track" command is given when
- the track buffer is modified, AND
- one of these:
  - the read/write head steps to another track, OR
  - the drive activity LED is (turns) off

This means it will be fairly unlikely that the bug is visible in normal use.
If the head steps to another track, any corruption to the track buffer will disappear a moment later when the data for the new track is loaded into the track buffer.
The LED is on as long as there is a file open. So to turn it off you need to close all files. Typically this means (I expect!) that the corresponding directory entry is updated first. Which makes us go to the other case (the head moves).
The most chance that this occurs is then on the directory track itself. The last byte of the last sector of track 18 is the HI byte of the number of blocks in some file. So we would notice if this number becomes larger than 255 blocks, for this particular file. For this particular file.
So, in normal usage, a rather small chance I would say.

If we really want to trigger this and verify that it happened, we would need to use a direct access buffer file, `open 2,8,2,"#"`. Then write some data into that buffer, 254 bytes to be exact. For example the values `02`...`FF`. That is easy to see if anything changed. Then using an `U2` command, write the buffer to the last sector of the track. To trigger the "save track", make sure the drive LED goes off, which most likely involve closing file 2 and possibly waiting a bit. At this point the buffer would get corrupted. Now we need to repeat the procedure to write to some other sector on the track and trigger the "save track" again. Now the corruption would be "safely" in the disk image file. It can be verified by checking the disk image. Or by yet again opening a direct access buffer to read the last sector of the track (`U1`), and compare all bytes with the data that was written into it.