sy2002/MiSTer2MEGA65 #53: error: type defaults to ‘int’

Rhialto, 2024-09-14:
I'm following the "first steps" tutorial and I ran into one very small and easily solvable snag:
```
loelappie:~/.../M2M/QNICE/tools$ ./make-toolchain.sh 
...
cc -I. -Icpus/qnice -Isyntax/std  -w -c -O2 -DOUTAOUT -DOUTBIN -DOUTELF -DOUTHUNK -DOUTSREC -DOUTTOS -DOU
TVOBJ cond.c -o obj/qnice_std_cond.o
cond.c:11:8: error: type defaults to ‘int’ in declaration of ‘ifnesting’ [-Wimplicit-int]
   11 | static ifnesting;
      |        ^~~~~~~~~
make: *** [make.rules:43: obj/qnice_std_cond.o] Error 1
cp: cannot stat 'vasmqnice_std': No such file or directory
```
Grep found me `./M2M/QNICE/c/vasm/cond.c:static ifnesting;` so I inserted `int` in there to fix the issue. I guess this is very very old C we're dealing with here...

sy2002, 2024-09-15:
Thank you @Rhialto ! We will fix this with the next release of the M2M framework.

sy2002, 2024-09-21:
FYI @Rhialto , I fixed this in the `develop` branch of the M2M framework which is a stable (yet not released) branch of the M2M framework. I propose you stay at the released V2.0.0 and use your manual fix. But if you decided to switch from V2.0.0 to our dev-branch then don't forget to apply a `git submodule update --remote --merge`, so that this very fix gets pulled from the QNICE repo.

Thanks again for reporting this - highly appreciated. 

I am closing this now as FIXED (and the fix will automatically be part of the next official release).

closed 2024-09-21 by sy2002