sy2002/MiSTer2MEGA65 #54: Joystick Port 2 doesn't work on MEGA R6

rjaremczak, 2024-11-06:
To observe wire-up flip joystick as described in "Hello World" tutorial and verify that only joystick port 1 is able to move the paddle. It can also be observed in BombJack port (tested on R6): after flipping joystick in OSM and trying to use joystick port 2 nothing happens.

rjaremczak, 2024-11-06:
C64MEGA65 core doesn't have this issue.

sy2002, 2024-11-06:
Thank you @rjaremczak - indeed the C64, the GameBoy and the ZX Spectrum don't have this issue, but if the demo core has this issue (on an R6 as you write): @MJoergen Can you please try to reproduce this on your R6 using M2M and the demo core?

@rjaremczak Which version of the M2M did you use? The released V2.0.0? The master branch? The develop branch?

rjaremczak, 2024-11-06:
My repo was generated from the template https://github.com/sy2002/MiSTer2MEGA65 so I guess it's 2.0.0 master branch

sy2002, 2024-11-06:
Oh, and @MJoergen : I do remember that you needed to activate a certain IC / power a certain IC in the C64 core due to a bug in the R4/R5 (R6?) PCB back in the days? An IC which then also powers this joystick port? I forgot the details - too long ago - was last year when we discussed that ;-)

rjaremczak, 2024-11-13:
Seems to be working fine on develop, closing.

rjaremczak, 2024-11-13:
leaving opened until verified and merged

sy2002, 2024-11-14:
Thank you @rjaremczak ! Highly appreciate your support! @MJoergen merged this to the "develop" branch. Can you switch your instance to the develop branch, rebuild and confirm that it works? (I unfortunately do not have a R6 machine to test.) If so, then let's close this one.

sy2002, 2024-11-14:
Sorry @rjaremczak : Do not switch to "develop" - it contains a lot of changes. We release a bugfix version 2.0.1 in the master branch soon.

rjaremczak, 2024-11-14:
Ok, perfect!

sy2002, 2025-04-28:
Released in V2.0.1

closed 2025-04-28 by sy2002