sy2002/MiSTer2MEGA65 #59: QNice sometimes hangs on MOVE @R5++, R12 after reset

Rhialto, 2025-03-29:
I'm just reporting this here to make sure that the details I researched so far get recorded.

In v0.000012 of the MegaPET core I changed the storage of the floppy disk images to HyperRAM. I suspect this issue is related. I mentioned it in the discord, so I'll just copy it here.

Sometimes when resetting the core, it doesn't reset properly, and sometimes (quite rarely) it doesn't even work when long-resetting it (until the power LED goes blue). I thought I had found some circumstances when this could happen, but it wasn't repeatable enough to debug so far. At first it seemed related to setting the 8296 memory to userport-control, and later
I thought it might be related to having a disk image inserted in the disk, but neither proved very repeatable.
    
For reference: when the reset doesn't complete, I have seen the QNice PC stuck at address `$1EFE`. That seems to correspond to this code. Meanwhile, the HELP key is unresponsive (I have learned that this usually means that QNice is busy with something else).
```
    007278                    ; Handle read request from drive number in R8:
    007279                    ;
    007280                    ; Transfer the data requested by the core from the linear disk image buffer
    007281                    ; to the internal buffer inside the core
    007282                    ;
    007283                    ; @TODO: Sanity check as described in vdrives.vhd protocol comment (3)
    007284  1ECE  FF90  005E  HANDLE_DRV_RD               ASUB enter, 1
    ...
    007314  1EF5  C600        _HDR_SEND_LOOP              CMP R6, R0                       ; transmission done?
    007315  1EF6  FFA3  0025                              RBRA _HDR_SEND_DONE, Z           ; yes
    007316
    007317  1EF8  0FA4  81DA                              MOVE VDRIVES_BUFS, R9            ; array of buf RAM device IDs
    007318  1EFA  1B24                                    ADD R11, R9                      ; select right ID for vdrive
    007319  1EFB  094D                                    MOVE @R9, @R3                    ; select device
    007320  1EFC  0111                                    MOVE R1, @R4                     ; select window in RAM
    007321  1EFD  05B0                                    MOVE @R5++, R12                  ; R12=next byte from disk img
    007322
    007323  1EFE  0FA0  7005                              MOVE VD_B_ADDR, R8               ; write buffer: address
    007324  1F00  0624                                    MOVE R6, R9
    007325  1F01  FFB0  2921                              RSUB VD_CAD_WRITE, 1
    007326
    007327  1F03  0FA0  7006                              MOVE VD_B_DOUT, R8               ; write buffer: data out
    007328  1F05  0C24                                    MOVE R12, R9
    007329  1F06  FFB0  291C                              RSUB VD_CAD_WRITE, 1
````
where I assume that the PC has already increased to the next instruction but the CPU is somehow stuck on the `MOVE @R5++, R12`.
    
So I'm getting a bit of a suspicion it is about the disk image which is now in Attic RAM.

Looking though the various vhdl files, there is the signal `s_qnice_wait_o` from each `entity work.qnice2hyperram` that eventually (if I traced it right) reaches `/M2M/vhdl/QNICE/qnice.vhd` where it is connected to `entity work.QNICE_CPU` as  `WAIT_FOR_DATA        => cpu_wait_for_data or ramrom_wait_i,`.

I suspect that the `MOVE` instruction is waiting for that (but I haven't put a 'scope on that to prove it yet).

As to why the `s_qnice_wait_o` might be active for "ever", I have no clue yet. Too much contention on the hyperram? The display meanwhile goes on fine, and so does the PET, as long as you don't try to access the disk.

Here is how I use the hyperram, which I hopefully correctly cribbed from the POC for the C64 core: https://github.com/Rhialto/PET_MEGA65/blob/v0.00012/CORE/vhdl/mega65.vhd#L681

MJoergen, 2025-03-29:
I think your analysis is sound, i.e. that the QNICE CPU is waiting for a memory access to complete. You can debug further by tracing the `hr_core_*` signals in mega65.vhd. Really only the write_o, read_o, and waitrequest_i signals are needed.

Rhialto, 2025-03-29:
I made some screenshots of the state after a reset where something hang. I added mark_debug to a few more signals than the ones you mentioned, in the hope that we can see where they are coming from.

 In the first case, not even the PET startup message appeared (which may mean it is a different case than what I'm interested in here). I have 2 pictures because there are 2 clocks involved but all these signals are stable at this time:

[image]

[image]

and here is a later attempt that did get to the startup screen, but the drive LED is on (this drive signals internal activity on the drive LED as well, I think) and the HELP key isn't functional.

[image]

[image]

sy2002, 2025-03-30:
@MJoergen and @Rhialto We never tried that QNICE accesses HyperRAM. This is "a first". Just to rule out "simple" things: QNICE is reading/writing from/to memory on falling edges and might it be that Michael's QNICE/HyperRAM RAM adaptor is acting/expecting rising edges and therefore we get lock-ups from time to time (maybe in cojunction with Clock-Domain-Crossing phenomenon)?

Rhialto, 2025-03-30:
For something that was never tried before it works very well indeed :) It is just failing on some edge case, so far.

Just some more ramblings.

Interesting note about the falling edge of the clock. While browsing I noticed in `M2M/vhdl/framework.vhd` that there is a comment `-- Clock domain crossing: QNICE to HR` using a `i_avm_fifo_qnice : entity work.avm_fifo` and `avm_fifo` which receives as clock `s_clk_i               => qnice_clk,`. It seems to end up using `~/tools/Xilinx/Vivado/2024.1/data/ip/xpm/xpm_fifo/hdl/xpm_fifo.sv` which only has `posedge` and never `negedge`, for both rd_clk and wr_clk.

There are similar `qnice2hr_d0_avm_fifo : entity work.avm_fifo` used in `CORE/vhdl/mega65.vhd` which receive `qnice_clk_i`. 

The `i_qnice2hyperram_d0 : entity work.qnice2hyperram` is also used there. That uses `falling_edge(clk_i)` internally so that seems ok.

So, if `avm_fifo` internally uses only `posedge`, but qnice acts on the falling edge of `qnice_clk_i`, then naively I think that inverting this clock when feeding it to avm_fifo would improve things. So I changed that in the avm_fifos in mega65.vhd (the ones attached to the disk images). That didn't immediately break the disk access, but after a few tries the problem could still be reproduced anyway.

```
   qnice2hr_d0_avm_fifo : entity work.avm_fifo
      generic map (
         G_WR_DEPTH     => 16,
         G_RD_DEPTH     => 16,
         G_FILL_SIZE    => 1,
         G_ADDRESS_SIZE => 32,
         G_DATA_SIZE    => 16
      )
      port map (
         s_clk_i               => not qnice_clk_i,	-- invert clock since qnice acts on falling edge
```
(and the same for `d1`)

I set mark_debug on some other wait request signals in i_framework around `i_avm_arbit_general : entity work.avm_arbit_general`. It seemed that while some of its outputs said "wait", its input didn't, but for some reason I did not get to see _all_ outputs. (hr_core_waitrequest_o might have been no-wait).

Meanwhile in mega65.vhd, we can see that `qnice_pet_mount0_buf_ram_wait` is 1 (coming out the qnice2hyperram) but the related input `qnice_disk0_waitrequest` is 0. Looking inside qnice2hyperram, this probably means that `reading` is 1 (since qnice_disk0_read and ..._write are both 0). It never got set to 0 which must mean that `m_avm_readdatavalid_i` never became 1. That would point upstream at the avm_fifo. Or it could be further upstream, such as the `avm_arbit`.

Here some more pictures, mostly because they have a few more signals on them.

[image]

[image]

MJoergen, 2025-03-30:
I'm on my way on vacation, so I can't help much. However, you might want to look in the branch "mfj_POC_disk_buffer_in_hyperram" in the C64 repo:

https://github.com/MJoergen/C64MEGA65/tree/mfj_POC_disk_buffer_in_hyperram

It's very old, but the name says it all. Maybe you'll find inspiration there.

Rhialto, 2025-04-06:
@MJoergen Thanks for the tip, that was indeed where I got my inspiration from,

I made some more pictures and added them below. It's as far as I got today but I'm not sure I can venture inside work.hyperram.

First a picture at reset of a normal, working, boot, so it's easier to spot what is missing when it fails. The signals are arranged more or less in order that goes upstream from the "drive 0 device" to wards the hyperram. Previously I had found that `i_qnice2hyperram_d0/reading` never turned off probably because it never saw `m_avm_readdatavalid_i`. I could not capture that signal for some reason but I have `m_avm_readdatavalid_d` which is one clock delayed, so good enough.
The extra signals I have now are further towards `work.hyperram` in `framework.vhd`. One can see that `hr_read` strobes (indicating the read request) and some time later `hr_readdatavalid` when the data is available. `m_avm_readdatavalid_d` pulses rather a "long time" later. (Would the fifo cause a long delay? I didn't research that)

[image]

Then one with a bit more contention on the hyperram but things still work fine.

[image]

Then a failure, with "barcode" on the screen, so here the hyperram got "stuck" even for the frame buffer. Note how `hr_read` pulses but `hr_readdatavalid` never responds.

[image]

and here a more normal failure (we do reach the normal PET boot screen), which actually seems to look totally the same:

[image]

MJoergen, 2025-04-08:
Just a random thought: Are you sure you're using the latest version of the M2M framework, i.e. V2.0.1 (same as master)? There was a fix in the HyperRAM controller a few months ago.

Rhialto, 2025-04-08:
I applied the patches from when that was first fixed, and then the patches from V2.0.1, but of course I can double-check if my end result is the same. Thanks for reminding me!

sy2002, 2025-04-09:
@Rhialto After applying the patch there must not be any "barcode" any more - otherwise we have a way bigger problem...

Rhialto, 2025-04-11:
I think I have a weak suspect: the `work.avm_arbit_general` in `framework.vhd`. I added most of the signals going into and coming out of it.

In two cases of "freeze", I saw the following sequence of events. 
- my trigger (which is still `CORE/i_qnice2hyperram_d0/reading` going up), 
- hr_core_read_o pulses, at the same time hr_address and a bunch of other hr_* signals change, and quickly after, hr_read pulses.
- from that I think that apparently the arbiter has decided to schedule the read from the core (from the disk image)
- before hr_readdatavalid becomes true, hr_dig_write becomes true, This is about 25 clocks later, though.

Either this sequence of events confuses the arbiter, or it is a coincidence (and things are stuck already when hr_dig_write becomes true). 25 clocks is a bit long for this time, but 15 clocks is not so unusual. 

I will attach some ILA exports. Screen shots are getting a bit difficult since there are too many signals to show.

[image]

[iladata.zip](https://github.com/user-attachments/files/19713409/iladata.zip)

Rhialto, 2025-04-11:
Maybe the "it is already stuck at that time" theory is true. In the ILA data where it is ok, I see read activity before "reading" (i.e. hr_readdata wiggles) but not in the case where it is stuck.

MJoergen, 2025-04-13:
@Rhialto Thank you for all your debugging effort. I'm confused (and quite troubled!) by your traces, since it seems the HyperRAM controller is not responding to the read request.

Could you please add debugging on the following signals inside `hyperram_ctrl.vhd`:
* `avm_read_i`
* `avm_waitrequest_o`
* `hb_read_o`
* `hb_dq_ie_i`
* `state`

Plus also the signal `ctrl_dq_ie` inside `hyperram_rx.vhd`.

This is inside the state machine of the HyperRAM controller. The protocol is that the read starts when `avm_read_i = 1` and `avm_waitrequest_o = 0`. The signals `hb_read_o` and `hb_dq_ie_i` connect to the HyperRAM device itself (through appropriate I/O stuff).

My suspicion (well, more like nightmare!) is that the signal `hb_read_o` gets asserted, but `hb_dq_ie_i` never does. That would indicate a problem in the communication between the HyperRAM controller and the HyperRAM device.

MJoergen, 2025-04-13:
One more comment. If the problem indeed is inside the HyperRAM controller, there is a timing parameter that can be tweaked. The value for `IDELAY_VALUE` in line 62 in `hyperram_rx.vhd`. You could try changing this parameter slightly, e.g plus or minus, say, five. This is purely trial-and-error. It's only useful if the error is reliably reproducible. But it could be nice to get some statistics, e.g. a range of values that work.
Note that you might get timing violations when changing this value. I would ignore those for the moment.

Rhialto, 2025-04-13:
I collected some traces with the signals you suggested. First some resets that went fine (1 and 2). Then some resets that got stuck with a black screen (3 and 4). A "long reset" didn't even help for these, so I re-programmed the fpga after nr 4. I have 16.384 samples (32.768 didn't fit) with the trigger at 12.000 ("reading" going up) but in the frozen case it seems like it's already stuck at that point since I see no read activity (and write activity could be very one-sided without response).

It seems that adding debugging to so many signals does affect the timing. With this core version, a good reset is much more rare.

I tried triggering on various reset signals, but the time before something interesting happens seems too long for the number of available samples.

I made a branch for this testing (debug-m2m-59), although the changes are mainly adding `mark_debug`s. I kept a qnice clock inversion into some `avm_fifo`s (based on the idea that those fifos internally work on rising clocks) but I never noticed any difference from that. Probably the signals stay stable long enough, several qnice clocks, so that it doesn't matter much.

I was thinking the problem would *not* be with the hyperram or its controller itself, since in many cases the screen output keeps working just fine while the disk image read is stuck. That would, I thought, point the suspicion more towards a fifo or an arbitrator or something like that. However with this core a stuck reset with working screen is also more rare.

Speaking of timing errors, there are a bunch listed in the logs, but most of them seem related to the ieee drive. In the C64 core, lots of those seem to be listed as exceptions, but for the MegaPET I have simply ignored them all (mentally). Or they were in the framework and I thought it would be not relevant to me. Maybe a bad assumption :)

Looking through the timing report I do see some mentions of hyperram. I will include the full report with the ILA traces. Hopefully it is not caused by the debugging (I don't have an old version around to compare, only one for R3 from the latest pre-release) because I can imaging that timing violations here could cause real problems.

Or it could be as simple that the state machinery isn't prepared for the case where `state == write_burst_st` while the read request comes in. At least that situation seems to be consistent in the failure cases I have seen, and doesn't occur in the success cases.

Traces collection nr 10 is from a random moment (no trigger used) while a reset has left the screen working, but the disk image read is stuck. There is lots of activity; it just seems like this particular read has been forgotten about and never gets serviced.

Sorry for the rambling, most of it is guesswork where I went from one guess to another. Hopefully the traces give some insight.

[iladata3.zip](https://github.com/user-attachments/files/19725521/iladata3.zip)

MJoergen, 2025-04-13:
I think you're on to something. The good news (for me!) is that the HyperRAM controller seems to work as intended:

The Avalon protocol is a bit "fragile" in the sense that if one initiates a burst write of 64 words, then the client is required to assert the "write" signal 64 times (while waitrequest = 0). If the client "misbehaves" then the Avalon bus (and hence the controller) gets "out of sync". This is seen in the traces, where the controller waits in the "write_burst_st" state. Essentially, the client has not given all the data it is supposed to. And unfortunately, the Avalon bus does not offer a way to get "un-stuck" or somehow abort an existing transaction. The good news is that bugs don't go unnoticed!

So one way to trigger on the event (or at least closer to the bug) is to trigger on the combination `hr_read = 1` and `state = write_burst_st`.

What we want to find out is, which client is not behaving correctly according to the Avalon bus requirements.

My biggest suspicion is indeed the `qnice2hyperram.vhd` file.

Note: I'm pretty sure it's correct to clear both `write` and `read` when `waitrequest` is 0. The reason is that when `waitrequest` becomes 0, then the transaction request indicated in the `write` and `read` signals is accepted.

Rhialto, 2025-04-13:
Thanks for the info! I made some traces with the trigger said like you said. Apparently it can happen that the trigger occurs but there is no hang. That seems to happen when a disk image is inserted. But in the case of a reset, it perfectly corresponded to a blocked read.

I'm not sure I can point at the "guilty party" from these traces. In general it seems that, when hr_read goes up, it corresponds to hr_dig_read going up. Who initiated the burst read before that, I could not really reliably see. It may be that something doesn't set hr_burstcount to a new value after a transaction finishes, or something like that, because I see no change in its value between the previous write and the current one (it remains 40).

[hrdebug4.zip](https://github.com/user-attachments/files/19727114/hrdebug4.zip)

MJoergen, 2025-04-13:
Since you mentioned timing violations, I ran a build (from tag v0.00012) without debug and indeed found many timing violations. I would strongly urge you to fix these violations before attempting any further debugging, because any timing violation can lead to obscure and random bugs.

Unfortunately, resolving all the timing violations is somewhat of a chore. Basically, one must consider each one in turn.

I can give you a head start by suggesting the following.

1. In the file `CORE.xdc` add the following two lines:
```
set_false_path -to [get_pins {CORE/i_main/ieee_drive_inst/bus_sync_o/s1_reg[*][*]/D}]
set_false_path -to [get_pins {CORE/i_main/ieee_drive_inst/bus_sync_o/s1_reg[*]/D}]
```
This is because the instance `bus_sync_o` is a dedicated Clock Domain Crossing, and hence timing violations here can be safely ignored.

2. In the file `pet2001hw.sv` perform the following changes:
```
@@ -152,8 +152,8 @@ dualport_2clk_ram #(
         .address_b(dma_addr[14:0]),
         .data_b(dma_din),
         .q_b(dma_rom_dout),
-        .wren_b(rom_wr),
-        .clock_b(dma_clk & ~dma_char_ce)
+        .wren_b(rom_wr & ~dma_char_ce),
+        .clock_b(dma_clk)
 );
 
 /////////////////////////////////////////////////////////////
@@ -181,8 +181,8 @@ dualport_2clk_ram #(
         .address_b(dma_addr[11:0]),
         .data_b(dma_din),
         .q_b(dma_char_dout),
-        .wren_b(chars_wr),
-        .clock_b(dma_clk & dma_char_ce)
+        .wren_b(chars_wr & dma_char_ce),
+        .clock_b(dma_clk)
 );
 
 //////////////////////////////////////////////////////////////
```

This is because one should avoid any logic directly on the clock pins.

3. In the file `ieee_drive.sv` there is an inconsistency between the comments in lines 60-62 and the block in lines 99-106. The latter is using the `clk_sys` clock, but the signals `img_size` belongs to the `clk_main` clock. I'm not sure what the best fix is, but perhaps you can add a clock domain crossing.

4. In the file `main.vhd` the signal `iec_drives_reset` belongs to `clk_main_i`, but the instance `ieee_drive` expects the `reset` signal to belong to `pet_clk_sd_i`. So I suggest adding a Clock Domain Crossing as follows in `main.vhd`:
```
@@ -546,9 +566,21 @@ begin
    --        This code currently only implements the "Always" option
    iec_drv_reset_gen : for i in 0 to G_VDNUM - 1 generate
       -- iec_drives_reset(i) <= (not reset_core_n) or (not vdrives_mounted(i));
-       iec_drives_reset(i) <= (not reset_core_n) or osm_i(C_MENU_UNIT_8_DISABLED); -- for now allow empty drives...
+       main_iec_drives_reset(i) <= (not reset_core_n) or osm_i(C_MENU_UNIT_8_DISABLED); -- for now allow empty drives...
    end generate iec_drv_reset_gen;
 
+   i_cdc_reset: xpm_cdc_array_single
+      generic map (
+         WIDTH => 25
+      )
+      port map (
+         src_clk  => clk_main_i,
+         src_in   => main_iec_drives_reset,
+         dest_clk => pet_clk_sd_i,
+         dest_out => iec_drives_reset
+      );
+
+
```

Rhialto, 2025-04-13:
Thanks! I will have something to do for the next weekend for sure! ;-)

MJoergen, 2025-04-13:
It might be that items 1 and 2 in my above comment is enough to allow us to proceed with the debugging. Because I just had a look at your latest traces, and `iladata-02-reset-hangs-black-screen` in particular looks suspicious.

This seems to be a trace right after reset, and I see that `hr_dig_write` is asserted and `hr_dig_waitrequest` goes to zero. This means the HDMI rescaler is writing to the HyperRAM, but it stops after 8 (or so) words (i.e. clock cycles). It should write 64 words.

I'm struggling to think why that could happen. The `hr_dig_write` signal is tied to `hr_write_o` in `digital_pipeline.vhd` and is generated by the file `avm_decrease.vhd`.

So if this particular failure is reproducible, then I would look at the internal state of the `avm_decrease.vhd` file, as well as its inputs.

Rhialto, 2025-04-15:
I am still in progress with trying to get rid of the timing violations. In the mean time a thought occurred to me. If the HDMI rescaler is behaving in an unexpected way, could that be caused by the core which outputs weird video signals? Because the PET certainly will do that, around reset time. The CRTC resets all its timing registers to zero, which surely will produce weird signals. It takes some time before the ROM code initializes it and that goes of course register-by-register, so that it takes a number of 6502 instructions before we get to the stable situation.

I realised I could test this by selecting a PET model without CRTC. Then its output is much more stable since in operation it just always follows the same timing which cannot be changed. Although while reset is asserted it will output nothing.

In this case it took 20 resets before it got stuck. That seemed longer than "usual" but on the other hand it wasn't a core with debugging enabled and those seem to make it worse.

I did notice that even then, sometimes there is a "barcode" effect for as long as I keep the reset button pressed (and it goes away when I release it). I don't know if that means anything, I didn't dig into the reset behaviour outside of the core.

Rhialto, 2025-04-18:
I have added traces on the internals of avm_decrease. Although it seems that triggering on `hr_read = 1` and `state = write_burst_st` perfectly corresponds to the blocking case, the actual blocking with `reading == R` happens much later. I didn't manage to catch both in the same window. But it is noticable that when triggering on the former, it doesn't get completely stuck yet, since "a long time" later there is more activity visible. So just in case it is helpful I have traces with both triggers.

avm_decrease writes in groups of 8. The outgoing burst count is 0x40, so it needs to write 8 groups of 8. s_write_pos cycles through 0...7.

Looking at e.g. traces set 03a-reset-black-screen, one can see inside avm_decrease that the trigger happens on the 9th group of 8. Just around the start of this 9th group, s_avm_write_i goes down. Then shortly after, s_avm_read_i goes up.  But this has only a delayed effect, once its state is idle_st. In between it still does a group of 8 writes, but once those are done, this stops. Since avm_burstcount is still at 0x40, this would be an incomplete burst.

So either the scaler wants to write but changes its mind before being finished with the write, or avm_decrease is too late with noticing that the scaler wanted to stop writing, and erroneously starts the next group of 8 writes. I'm not sure which is which, what is cause and what is effect here.

[hrdebug5.zip](https://github.com/user-attachments/files/19813498/hrdebug5.zip)

MJoergen, 2025-04-21:
I see you've fixed the timing violations. Well done!

From your observations, I thought the bug could perhaps be a problem with reset.

Anyway, I tried building using the branch `rhialto`, i.e. commit `f34eff7`. However, the output is just a black screen. So I did some debugging and I see the signal `main_qnice_reset` in the file `top_mega65-r6.vhd` is always asserted. This will force both the core and the M2M framework into reset, hence the black screen.

Furthermore, when using the serial port (UART) and do see output, so the QNICE is started up. The last it writes is `Config file not found: /pet/petcfg  Using factory defaults.`

Anyway, I don't know if this is related to your problems.

Does your core start up, i.e. show video output on VGA? Do you have any local changes not committed, for this to work? Are you on R3 or R6? I'm testing on R6 at the moment.

MJoergen, 2025-04-21:
Oh, and a quick comment on your latest traces (03a-reset-black-screen): I notice in clock cycle 8118 that `hr_address` changes from 0x11f00 to 0x11f40. This suggest to me that a new transaction is about to start. However, this seems to be right in the middle of a lot of trafic. That's why I'm guessing that some parts of the design are being reset, while other parts are not.
However, looking briefly at the source code, I see nothing wrong.

Rhialto, 2025-04-21:
Commit `f34eff7` is the right one, I just checked. I will adjust the debugging branch shortly to include my extra mark_debugs. 

Do you mean the output is *always* a black screen, or only sometimes? Because for me it works most of the time and fails once in every 5-10 times or so. But I only use IMDH, not the VGA output. I seem to remember that the VGA output is fairly unprocessed as it comes from the core, and PET video output is not very VGA-like. It can change timing, too, because the CRTC is programmable. I had some reports from people trying to use the VGA output and those say it is very bad and/or doesn't work. I guess it depends on the flexibility of their monitors. Different PET models had different CRTs with different timing requirements... for whatever reason.

I am using R6, I don't have an R3. From what I heard, the R3 version of the framework did a nice job of making a version for that board.

The core should be able to work without any files on the sd card. All ROMs are pre-set into the core. The preferences file is also not required.

MJoergen, 2025-04-21:
Ok, I got the core working. My problem was I was viewing the VGA output and therefore did not see the Welcome screen, and it didn't occur to me to press the space bar ...

sy2002, 2025-04-23:
@MJoergen Out of curiousity: We are only showing the welcome screen on HDMI? I always thought that the HDMI and VGA output are in sync?

Rhialto, 2025-04-23:
I think it is more likely that the vga monitor doesn't like the signals the MegaPET puts out. It is one of the things I have postponed to some future date: can the vga output be more like vga without compromising the cycle-exactness of the video generation. Another is: can I change the "hdmi zoom" parameters so that it zooms in to the screen area a PET really uses (which is at least 3 sizes: 3032 without crtc, 4032 in text mode, 4032 in graphics mode). And there are demos that coax the crtc in doing something like 90 chars per line and 33 lines, or something like that, but as far as I'm concerned the zoom doesn't need to handle such outliers.

sy2002, 2025-04-23:
> can the vga output be more like vga without compromising the cycle-exactness of the video generation.

I did not look into your code and therefore this answer is super high-level and so not sure if helpful at all: MiSTer offers a few tools like a scandoubler and other things that we moved into the M2M framework, you can use them, too: https://github.com/sy2002/MiSTer2MEGA65/tree/master/M2M/vhdl/controllers/MiSTer

Here is some documentation: https://mister-devel.github.io/MkDocs_MiSTer/developer/video_mixer/#instantiating-video_mixer

You can also write into a frame buffer and use another M2M component for that:

https://github.com/sy2002/MiSTer2MEGA65/blob/master/M2M/vhdl/av_pipeline/frame_buffer.vhd

This framebuffer is able to generate a clean VGA signal.

Rhialto, 2025-04-23:
Thanks, that looks interesting (even if it turns out that it doesn't perfectly fits the situation).

MJoergen, 2025-04-23:
@sy2002 

> Out of curiousity: We are only showing the welcome screen on HDMI? I always thought that the HDMI and VGA output are in sync?

Well, the core is held in reset during the welcome screen. I checked on the C64 core (enabling the welcome screen) and there it is shown on both VGA and HDMI, even though the core is held in reset.

So I'm guessing the PET core is simply not generating any useful video signal when held in reset.

MJoergen, 2025-04-23:
> I think it is more likely that the vga monitor doesn't like the signals the MegaPET puts out. It is one of the things I have postponed to some future date: can the vga output be more like vga [...]

A place to start is to play with the values of `VGA_DX` and `VGA_DY` in the file `CORE/vhdl/globals.vhd`.

MJoergen, 2025-04-25:
@Rhialto Ok, I've found something you can try. In the file `M2M/vhdl/av_pipeline/ascal.vhd` uncomment the lines 1135 and 1647. These two lines I (apparently?) commented out a few years ago, but I don't have the faintest clue why.

But I think changing these two lines will fix your reset problems.

sy2002, 2025-04-25:
@MJoergen Well, you commented this one out because we had sporadic / erratic HDMI reset problems back in the days 🙂  Let's be careful when it comes to changing stable M2M code (outside of the PET project of course. Inside a project, one can of course do what one wants ;-) )

MJoergen, 2025-04-25:
> [@MJoergen](https://github.com/MJoergen) Well, you commented this one out because we had sporadic / erratic HDMI reset problems back in the days 🙂 Let's be careful when it comes to changing stable [...]

Absolutely. I agree!

However, I would like @Rhialto to try this test (uncomment the two lines mentioned) as a test/experiment, to see whether it changes anything (hopefully for the better).

I do vividly remember the problems we had chasing down various bugs related to HyperRAM, HDMI, and reset!!

The change suggested is reverting part of a larger refactoring done in the C64 repo in commit 15cec30dd on 2023-05-23 and in the M2M repo in commit 4c9492d3 on 2023-05-22.

If @Rhialto sees improvement, then we need to think hard about what to do. For now, I propose we don't do anything, except let @Rhialto enjoy it. Later we may consider experimenting with this change in the `develop` branch of the M2M repo.

Rhialto, 2025-04-25:
@MJoergen @sy2002 Yes! That change really helps! I have now done about 33 resets and not a single time was there a problem. Thank you very much!

So it looks like you indeed will have to consider what to do with these lines in the Framework in the long run... sorry about that.

I think I can now start working on the other known weak point, the 4040/8250 disk drive. (I reported upstream and they said it was related to the fact that the DOS sometimes quickly switches between the 2 drives for a brief moment. Both drives share a single memory buffer to store the current track. That always seemed like a false economy to me. I want to try to give each drive its own buffer. Especially since the drive seems to work less reliably than I thought/noticed before. Now even a LOAD"$",8 (which accesses both drives) on the 4040 seems to read incorrect data from the disk images...)

sy2002, 2025-04-26:
@Rhialto I was not following along this thread due to being very busy. Can you please summarize for me: Did Michael's suggestion to comment-in these two very lines actually fix the "QNice sometimes hangs on MOVE @R5++, R12" issue, i.e. the issue that if QNICE works with HyperRAM instead of static FPGA RAM the system sometimes locked up? (In other words, this issue could be closed?)

Or did you / will you need to fix more things to fix this very issue?

I am asking because we of course need to decide if we implement this fix then upstream into M2M @MJoergen . Your guidance about how "dangerous" this is, would be needed. We could also park it and create an issue to think about it at a later stage.

Rhialto, 2025-04-26:
@sy2002 I can confirm that the suggested change works. I didn't do an infinity of testing but at something like 35 resets there was not a single instance of this issue. Before it would manifest in roughly 5-10 resets. I did not fix anything else. The commit I made is https://github.com/Rhialto/PET_MEGA65/commit/f1ce8dd08cc5592d102dd21a897c9956719a6d28 .

I have been thinking about what makes this core different from other cores. I think it was mentioned that it produces no video signal while the core is in reset. It could be related. Unfortunately one cannot expect a proper video signal from the CRTC before the ROM code initializes it. I could in principle switch to using the pre-crtc style hardware while the core is in reset. It is there to support the 2001 and 3032 models. But that would still leave a short time between when reset ends and when the 6502 gets around to initializing the CRTC. I'm not sure if that would make enough of a difference to be worthwhile.

Rhialto, 2025-07-07:
> [@Rhialto](https://github.com/Rhialto) Ok, I've found something you can try. In the file `M2M/vhdl/av_pipeline/ascal.vhd` uncomment the lines 1135 and 1647. These two lines I (apparently?) commented [...]

I think I can see one reason why these lines were commented out. Now that they are re-enabled, there are more "unwanted" things visible (at least on the hdmi display) at startup time. It is probably un-initialized memory which gets displayed as a barcode pattern. Lots of vertical stripes in random colours. It takes a second or so, then everything is properly initialized and the display is clean. It is not horrible to look at in any way but all those colours look a bit unexpected when booting into a (mostly) monochrome core :-)

But the change works perfectly, I never had this issue again. So I'm not complaining.